macdylan/3dp-configs #1: Question: do copy/mirror/backup modes work?

Googliola, 2023-08-24:
Found this gem in the start gcode section:

{if physical_printer_preset =~/.*IDEXDupl.*/ || physical_printer_preset =~/.*IDEXCopy.*/ }
  M605 S2 X162 R0 ;IDEX Duplication
{elsif physical_printer_preset =~/.*IDEXMirr.*/}
  M605 S3 ;IDEX Mirror
{elsif physical_printer_preset =~/.*IDEXBack.*/}
  M605 S4 ;IDEX Backup
{endif}

Does it work? and if so, how can it be enabled?

That capability would be a huge improvement!

Thanks a lot for your hard work here!

macdylan, 2023-08-24:
Add one or more physical printers with these keywords(IDEXMirror/Copy) in their names.
[image]

In OrcaSlicer, include these keywords in the name of the Plate settings:
[image]

Googliola, 2023-08-24:
Thanks a lot! Working now.

Some notes:

1. Turns out you need to restart Prusaslicer after creating the new Physical Printer or it wont be picked up in the gcode.
2. If I change from IDEXCopy to IDEXMirror, the gcode does NOT get re-sliced and the Physical Printer will still be the same (IDEXCopy in this case, check the gcode). Probably related to note 1. PrusaSlicer 2.6.0
3.

closed 2023-08-24 by Googliola