macdylan/3dp-configs #2: Copy / Duplication mode not working

Googliola, 2023-08-24:
How to reproduce (in PrusaSlicer ; gcode attached):
1. create simple shape on the left side of the plate
2. select physical printer with IDEXCopy or IDEXDuplicate in the name
3. Export to Octoprint and print it
-> only left item gets printed

4. Use same shape but export in "normal" (single) mode 
5. Print from USB stick in Copy Mode
-> two copies

6. Now go back to Octoprint and print the very same gcode from step 3
-> tadaaaa. duplication is ON!

In order to narrow it down, I checked Marlin documentation and found 
_2 : (DUPLICATION) [....] Set the constant X-offset and temperature differential with M605 S2 X[offs] R[deg] and
        **follow** with M605 S2 to initiate duplicated movement._

I was hoping that
```
M605 S2 X162 R0 ;IDEX Duplication
M605 S2;  
```
makes a difference as it says _follow to initiate_. It did not.

I have no clue why that happens. 

[Copy Test Gcodes.zip](https://github.com/macdylan/3dp-configs/files/12431138/Copy.Test.Gcodes.zip)

macdylan, 2023-08-24:
Let me explain how J1 works in Copy/Mirror/Backup mode:
- Read `;Extruder Mode:`([values](https://github.com/macdylan/SMFix/blob/84ef11e91c2daca59de5125556b8470594968c29/fix/const.go#L8)) in headers to display that mode on the touchscreen, but just "display".
- At the beginning of printing, the controller will execute `M605 xxx` if it exists.

so, I made the start gcode(#1 ):
```
{if physical_printer_preset =~/.*IDEXDupl.*/ || physical_printer_preset =~/.*IDEXCopy.*/ }
  M605 S2 X162 R0 ;IDEX Duplication
{elsif physical_printer_preset =~/.*IDEXMirr.*/}
  M605 S3 ;IDEX Mirror
{elsif physical_printer_preset =~/.*IDEXBack.*/}
  M605 S4 ;IDEX Backup
{endif}
```

and check `M605 xxx` in smfix to set `Extruder Mode`. so in this way the screen and the actual gcode command are the same.

Let's come back to topic, for some strange reasons in PrusaSlicer, something might not work when switching the physical printer. I recommend select the mode first(physical printer), then set other options, and finally slicing. But in OrcaSlicer, there is no such problem, you just need to set the plate name arbitrarily.

Feel free to ask that you are not clear about.

macdylan, 2023-08-25:
> Long story short: copy mode exclusively works when started from either the touchscreen or Luban. But not in Octoprint. Mirror works; backup, I dunno.

That's weird, but copy mode works fine on my J1 and I use it a lot, fw 2.6.2/ps 2.6.0/orca 1.6.4/sm2up 2.2(smfix 1.9)
[CopyTest-Box_0.2mm_27m_J1.zip](https://github.com/macdylan/3dp-configs/files/12434780/CopyTest-Box_0.2mm_27m_J1.zip)

Googliola, 2023-08-25:
Uploaded your test files to octoprint, did not work. Like I said, since mirror mode works, I suspect that the parameters S2 X162 get lost somewhere...Hopefully, SM reads and fix the issues on their github repo in the near future. Probably wishful thinking.

closed 2023-08-25 by Googliola

macdylan, 2023-08-25:
I rarely use OctoPi and usually for testing purposes. but i just linked to it and successfully tried the Copy mode:
- use J1's touchscreen to complete a Copy mode file. this step will activate T1, but in `M605 S2` you can't do that.
- use OctoPi to print other gcode files, everything looks OK.
[1-active-x-mode.zip](https://github.com/macdylan/3dp-configs/files/12440839/1-active-x-mode.zip)

I think the touchscreen is doing some black box things, from the debug log you can see it activates T1 after M605 S2 but I'm not sure exactly how that works yet, I'll keep digging it.

Googliola, 2023-08-25:
Don't we love black boxes!

Found these lines in Marlin / Snapmaker-IDEX repo:

   * M605: Set multi-nozzle duplication mode
   *
   *  S2       - Enable duplication mode
   *  P[mask]  - Bit-mask of nozzles to include in the duplication set.
   *             A value of 0 disables duplication.
   *  E[index] - Last nozzle index to include in the duplication set.
   *             A value of 0 disables duplication.

No idea if that is related to the sneaky T1.  Happy digging ;-)

TylonHH, 2024-05-13:
@Googliola Hey hey,
as you guys figured out its not a slicer problem, then more a black box.
I just wanna share my workaround, when I wanna print in **copy** mode via **OctoPrint**. I saved a gcode like "SET copy.gcode" on a stick including this fast code:
```T0
M82 ;absolute extrusion mode
;--- Start G-code Begin ---
M605 S3
;M605 S2 X162 R0
G28 Z
M84
;--- End G-code End ---
M82 ;absolute extrusion mode
M104 S0
;End of Gcode
```

I saved this code on the printer internal storage for easy access and setting it when needed.
After running this snippet, you can print in copy mode via OctoPrint.