macdylan/3dp-configs #3: Orca Configs for Snapmaker J1 have long purges/retractions

KyGunsAndRadio, 2024-02-02:
Downloaded latest OrcaSlicer v1.9.0. I noticed this release was supposed to include your latest profiles, which is great. But, the start/end gcode contain multiple long purges and a long retraction at the end. These aren't needed for an IDEX printer. The purges just make a mess in the printer, but the retraction jammed up my extruder gears on the very first print with this version. These were not present in previous versions.

Sure it's easy enough to comment them out, but is there a reason these were added that I'm not considering?

[image]

KyGunsAndRadio, 2024-02-02:
So, I did find your explanation of the changes in the readme. I can understand the purges, but in my case the retraction definitely caused issues. Also, for some reason the right nozzle didn't return back to the silicone stop after the initial purge. I reverted back to your 20230731 gcode as it worked perfectly for me.

le-avion, 2024-02-04:
I am having the same VERY long purge before the print begins. This is on the Prusaslicer. The readme talks about 4-8 cm but what I am getting is more like a meter. Not sure how to fix it. Where is this defined?

KyGunsAndRadio, 2024-02-10:
@macdylan Thanks for the followup and for all your work on Snapmaker printers. My J1 would probably not get used if it wasn't for your profiles and sm2uploader.

@le-avion The purges are coded directly in the Start G-code. Look under the printer settings and Machine Gcode. Make sure advanced settings are enabled. You will see a few lines like G1 E50 (may not be exact). Those are the purges. The retraction is in the End Gcode.

macdylan, 2024-04-29:
Hey guys, a new version has been released:
https://github.com/macdylan/3dp-configs/releases/tag/20240429_6b549a3

I've been testing them for a while, please give them a try:
[image]
[image]

closed 2024-05-15 by macdylan