macdylan/3dp-configs #4: Snapmaker A350 has ridiculous start GCode setting temp to min 250C for all filaments

tyeth, 2024-05-06:
[Isn't this madness?](https://github.com/macdylan/3dp-configs/blob/main/PrusaSlicer/Snapmaker.ini#L3598), the original A350 was rated at 250c Max, and the start gcode picks the 
max of (250 or min(290 or (firstlayerTemp + 15))

That's likely to cause overcooked filaments for those not tolerating 250c.

Then the start gcode also set the temp to initial layer + 5, equally that feels like a machine specific choice unless we are just accepting the snapmaker is off by 5 (probably fair).

Surely if the user specifies a temp for a first layer it should be adhered to as maximum during start gcode.

macdylan, 2024-05-07:
Max nozzle temperature of A350 is 275C, but 290 is OK because the fw will limit it.

The temperature is set in the filament params, I only set the same temperature for compatibility with Dual. And after testing, this temperature setting does not cause the filament to overcooke. If you find some problems, please let me know the brand and model of filament.

tyeth, 2024-05-07:
latest firmware I've installed allows me now to select up to 300c, but no ability to label which print head version is installed, so I just stick to 250max. Snapmaker are not known for keeping firmware compatible with older models, as they are always keen to push latest features (move fast break things).

I think around 2021 they redesigned the cooling for the single print head. Sadly I have the older model, but the newer one was supplied with all the AT/F models and maybe newer purchases of A models (as well as separate purchase).
https://forum.snapmaker.com/t/new-3d-printhead-its-redesigned/14992/17

macdylan, 2024-05-07:
I understand the difference between old & new. However, in the preparation stage, the temperature of the nozzle is raised and then purge, that time is very short(4-6 seconds), so I don't think it will cause clog. Maybe I'm wrong?

tyeth, 2024-05-07:
I think I saw it always attempting to reach 250 (takes mine a while if bed
heating too) and got scared how long it would be there. I also had other
blockage issues and just changed the nozzle so temporarily more risk
avoiding, along with using much lower melt point filaments than my usual
petg at 245. I've also singed the ptfe tube in one hotend while running at
the 250 mark in the past (probably crap tolerance filament jam related)

All anecdotal rubbish, so take it with a pinch of salt, I guess I was
worried this profile was now effectively for the dual heads temp ranges.