macdylan/3dp-configs #5: J1: How to print Dual Color

TylonHH, 2024-05-13:
I wanna print via OctoPrint with two filaments and slice with Orca. So I downloaded the latest config here and replaced as mentioned.
I wonder why in the whole gcode is no M605, even no
```
;M605 S0 ;Full control mode
;M605 S1 ;Auto-park mode
```
anyways: I checked the gcode I and see the fast switchung tool comand M2000. But at the 1st switch the T0 stays where it is, and the T1 cant print where it should. Is this because of OctoPrint?

macdylan, 2024-05-14:
I stopped trying to use OctoPrint and J1 together until the snapmaker team solve M605 issues.

J1 works in M605 S0 by default. S1 also works fine, but I will continue to use S0 mode with M2000, worked very well.

TylonHH, 2024-05-24:
You're right. Printing via USB works fine.