macdylan/3dp-configs #6: A350 startup gcode seems overengineered?

chege54, 2024-05-22:
https://github.com/macdylan/3dp-configs/blob/61e20e9a56681def4d6c4bb0643f4dbc827cf7e6/PrusaSlicer/Snapmaker.ini#L3398

I would have a few question regarding to the A350 startup code:
- Why the head moved to the "center" of the bed?  
`G0 X{if initial_extruder == 0}145.0{else}175.0{endif} F3420.0` in my opinion it would be better outside `G0 X{if initial_extruder == 0}-5.0{else}...`

- What is the meaning of `C2 W1;` and `C3 W1`?
`M109 S{max(250, min(290, first_layer_temperature[initial_extruder] + 15))} C2 W1; common flush temp`
...
`M109 S{first_layer_temperature[initial_extruder]} C3 W1`

- My experience is with this part:  
`M104 S{first_layer_temperature[initial_extruder] + 5}`
`G1 E9.0 F200`
`G92 E0`
`G1 E6.0 Z4.6 F200`
`G92 E0`
`G0 Z5.6 F200`
extruder is standing in one place and extruding, the melted filament is backfireing on the nozzle and gets clogged. Like so:
[image]
What is the use case of it?

macdylan, 2024-05-23:
> * Why the head moved to the "center" of the bed?

During nozzle warmup, it is easier to manually clean the nozzle and load the filament by moving to the central position. Especially when using enclosure.

> * What is the meaning of C2 W1; and C3 W1?

https://github.com/Snapmaker/Snapmaker2-Controller/blob/7c88979c4f10d65c8950440fdf122ebb22bfd920/Marlin/src/gcode/temperature/M104_M109.cpp#L140

> * ... The melted filament is backfiring on the nozzle and it gets clogged...

Your nozzle looks slightly blocked, please replace it with a new one. or the filament is wet :)

chege54, 2024-05-23:
> During nozzle warmup, it is easier to manually clean the nozzle and load the filament by moving to the central position. Especially when using enclosure. [...]

But the head is at Z1.6, which is too close to the print bed. How can you clean it? I can not access/reach it.

I dried the filament for 24 hours in a sunlu S2 drier, assume it is (totally) dry, I was able to print it without any issues from a luban generated code. Nozzle is blocked because print head it stayed in one position and extruded ~20mm filament under itself. (That is problem that I wanted to highlight)

Btw. Thank you for the C and W explanation. I did not found it in the docs. But it is documented via code 🙂

macdylan, 2024-05-23:
> But the head is at Z1.6, which is too close to the print bed. How can you clean it? I can not access/reach it.

Ohh, I read it wrong. I thought it was G0 Z216, but on Z1.6 it would look like this:
[image]

chege54, 2024-05-23:
> > But the head is at Z1.6, which is too close to the print bed. How can you clean it? I can not access/reach it.
> 
> 
> 
> Ohh, I read it wrong. I thought it was G0 Z216, but on Z1.6 it would look like this:
> 
> [image]
> 
> 
> 
> 

Exactly

jakubkinst, 2024-07-10:
I am experiencing the same problem. Since the purge is happening super-close to the build-plate, the nozzle actually gets clogged and dirty instead of the opposite. @macdylan can you please fix this part? Thanks

chege54, 2024-07-17:
> I am experiencing the same problem. Since the purge is happening super-close to the build-plate, the nozzle actually gets clogged and dirty instead of the opposite. @macdylan can you please fix this part? Thanks

thanks for confirming. I will try to solve it at the weekend